MOSAIC-LoPoW/octa-stack-students #15: Unable to call LoRaWAN_send() in interrupt callback

SamGeens, 2021-11-09:
If I execute the LoRaWAN_send() method in an interrupt callback the Octa crashes with: `[ERR] ASSERT in file: ../../../core/ST/Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F/port.c on line: 415`. 

To reproduce the error I used the standard `murata-dualstack-example` and added a button interrupt. In the interrupt callback I tried to send a Lora message. I left the timed Lora send in the code for testing purposes. As long as the interrupt is not called everything is working fine and the messages are sent and even received in TTN. But as soon as I call the button interrupt the Octa crashes.

```
void Test_Appcallback(void)
{
  printINF("Execute interrupt callback\r\n");
  LoRaWAN_send(NULL);
}
```

**The complete example application is added as a .zip file:**
[button-interrupt-lora.zip](https://github.com/MOSAIC-LoPoW/octa-stack-students/files/7508040/button-interrupt-lora.zip)

SamGeens, 2021-11-09:
Using a flag in the interrupt handler and checking the flag in `void murata_process_rx_response(void const *argument)` seems to be working correctly.

**Interrupt handler:**
```
void Test_Appcallback(void)
{
  printINF("Execute interrupt callback\r\n");
  sending = 1;
}
```

**Check flag in murata_process_rx_response:**
```
if(sending){
    LoRaWAN_send(NULL);
    sending = 0;
  }
```

Even though this seems to be a working solution I'm still curious if this is an optimal solution aswell?

Klievan, 2021-11-09:
Hi Sam!

In the `void LoRaWAN_send(void const *argument)` function, the CMOSIS function `osMutexWait(txMutexId, osWaitForever);` is used to 'lock' the application to prevent other technologies from sending while transmitting over LoRaWAN. This function can't be called from an ISR (interrupt service routine) as can be read on the page below:
https://www.keil.com/pack/doc/CMSIS/RTOS/html/group__CMSIS__RTOS__MutexMgmt.html#ga5e1752b73f573ee015dbd9ef1edaba13

Looking at your example, you're not using any other technologies such as NB-IoT, Dash7 or BLE to send data so I'm not sure if that's needed in this case. I've only started recently working with the OCTA-stack.

You can call the LoRaWAN_send function from the main application context, and using a flag is an excellent way to achieve this. Generally it's a bad idea to perform 'heavy' actions such as sending/storing data directly from an interrupt-context, as this often can be done on the main thread. Otherwise you might be holding up other pending interrupts from the same or lower priority.

How these issues often are solved in embedded development is the creation of an action queue which is executed in the main loop, while it gets filled with actions from interrupts which need to perform tasks which should not be performed in interrupt context. For your project this seems overkill though and flag like yours is perfectly fine. Maybe Mats has some extra comments on this?

MatsDeMeyer, 2021-11-09:
Fully agree with that Ivan says.

The assert happens for a reason, which is explained in the comments surrounding it (line 415 in the port.c file)

```c
void vPortEnterCritical( void )
{
	portDISABLE_INTERRUPTS();
	uxCriticalNesting++;

	/* This is not the interrupt safe version of the enter critical function so
	assert() if it is being called from an interrupt context.  Only API
	functions that end in "FromISR" can be used in an interrupt.  Only assert if
	the critical nesting count is 1 to protect against recursive calls if the
	assert function also uses a critical section. */
	if( uxCriticalNesting == 1 )
	{
		configASSERT( ( portNVIC_INT_CTRL_REG & portVECTACTIVE_MASK ) == 0 );
	}
}
/*-----------------------------------------------------------*/
```

The general idea is to do as little as possible from the interrupt callback.
Your solution with a simple flag is what I would recommend to use!

SamGeens, 2021-11-15:
Thank you both for the extra information! Feel free to close this issue.

closed 2021-11-15 by MatsDeMeyer