MOSAIC-LoPoW/octa-stack-students #16: OCTA BLE shield

SamGeens, 2021-11-16:
Is it possible to receive a OCTA BLE shield? (preferably 4 shields)

MatsDeMeyer, 2021-11-16:
There is already a BLE module on the OCTA, no shield needed.

Check the pinout section of the wiki for details.

jorredebacker, 2021-11-16:
Why did we used another board to work with bluetooth? If we could use the octa shield?

MatsDeMeyer, 2021-11-16:
Easier to work with, the example for the development kit work out of the box.

The OCTA shield needs some configuration for it to work, for example its UART is only connected to the OCTA and not to a USB port.

Klievan, 2021-11-16:
The nRF on the octa is the nRF52832, so if starting from an example make sure you're using the pca10040 and s132 versions of the example. Depending on your project, you can start from the example that lies the closest to your use-case. 

Like I explained in the lessons, if you need a phone to connect to your device, go for a peripheral example (phone initiates connection). If you need your device to look for other devices, or initiate a connection, go for a central example. If you need both, go for a "BLE Central & Peripheral" example. 

The most barebones example is the "template example" which is stripped the most, but you'll have to add your own services to that. Below a tutorial on how to get started with creating your own custom service:

https://github.com/NordicPlayground/nRF5x-custom-ble-service-tutorial

SamGeens, 2021-11-16:
For those who also have trouble running the `template_project`: [See this](https://github.com/makerdiary/nrf52-u2f/issues/14)

SamGeens, 2021-11-16:
If I try to run the same project on the Octa I get the following:
[image]

Any ideas?

Klievan, 2021-11-16:
> If I try to run the same project on the Octa I get the following: [image]
> 
> Any ideas?

99% that you are trying to connect to the wrong debug port. You need to use the cortex connector that is labeled BLE/nRF52 together with a J-Link programmer, or a nRF52 devkit in programming mode.

closed 2021-11-28 by SamGeens