MOSAIC-LoPoW/octa-stack-students #18: Windshield Frost Detector

TomRaeves, 2021-11-16:
We were wondering if we could use this sensor. For our project we would like measure the temperature and humidity on the outside of the car. We found the following sensor because it is waterproof and can be connected to either the lorawan network or Bluetooth. If we are allowed to use this sensor, we were wondering if we can directly send the data to our backend via lorawan. We will be using the octa-stack for measuring the inside temperature/humidity inside the car. Are we also required to use the octa-stack to connect to this outside sensor because then we would probably send the data via bluetooth to the octa-stack. 

The link to the sensor:

https://www.lairdconnect.com/wireless-modules/lorawan-solutions/sentrius-rs1xx-lora-enabled-sensors?matchtype=b&sncid=13&utm_source=google&utm_medium=cpc&utm_campaign=platforms_general&adgroup=rs1xx&gclid=Cj0KCQiAys2MBhDOARIsAFf1D1eYvJKlgLZpCkTmNUnRsrGfzwAtAMJkKhspNycgg4ow8p0E-mQSVHsaAgwMEALw_wcB

MatsDeMeyer, 2021-11-16:
I think going for a real waterproof setup is a bit overkill. 
Your project will remain a proof of concept, so buying a purpose built waterproof sensor just to give some sort of demo in the end might be a bit much.

Or are there other features this sensor has that the octa does not?

TomRaeves, 2021-11-16:
no it does not have additional features but we need to measure the indoor and outdoor temperature/humidity at the same time. We could also use 2 octa-stacks if this is more convenient ?

Klievan, 2021-11-16:
Are you 100% certain that just getting the temperature and humidity will ensure that the car-window is going to be frosted? Just brainstorming here, but would measuring the backscatter of a light on the inside of the car give you an indication wether the windshield is clear or not?

closed 2021-12-03 by MatsDeMeyer