MOSAIC-LoPoW/octa-stack-students #21: multiple defenition error

thibeaudemol, 2021-11-16:
https://github.com/thibeaudemol/IOT_testing

[image]

MatsDeMeyer, 2021-11-16:
Had a look at your code there, the issue is in your header file of acc-mag-sensor.

This seems to be two header files pasted below each other. By cleaning this up, I managed to compile the code, however `lightsensor_measurement()` is not defined in your source file.

thibeaudemol, 2021-11-16:
My header now looks like this, but it still doesn't compile. 

[image]

MatsDeMeyer, 2021-11-16:
I have this as a header file:

```c
#include "platform.h"

osTimerId accelerometer_timer_id;
osTimerId lightsensor_timer_id;
osMutexId i2cMutexId;
osThreadId defaultTaskHandle;
osTimerId iwdgTimId;

uint8_t count=0;

struct RGBvalues
{
  /* data */
  int red, green, blue;
};

void StartDefaultTask(void const *argument);
void accelerometer_measurement(void);
void print_accelerometer(uint16_t data[]);
void lightsensor_measurement(void);
void RGBsensor_measurement(void);
void print_light_intensity(int data);
void print_color(struct RGBvalues values);
```

The only issue I have is the undefined reference to lightsensor measurement (which makes sense as it is not declared anywhere)

```c
/Downloads/IOT_testing/applications/examples/acc-mag-sensor/../../../applications/examples/acc-mag-sensor/src/acc-mag-sensor.c:55: undefined reference to `lightsensor_measurement'
collect2: error: ld returned 1 exit status
```

MatsDeMeyer, 2021-11-16:
fyi, you give the count variable a value in the header file which I would not recommend

closed 2021-11-16 by thibeaudemol