MOSAIC-LoPoW/octa-stack-students #25: OCTA BLE driver

TissieVA, 2021-11-23:
Is there a driver for the Bluetooth of octa. When I looked at the example of the temperature sensor, it had an included "sht31.h", the name of the component. The same for the battery example but with STC3115. 

I was now trying to use BLE component of our octa named ISP1507, but I can't seem to find a driver file "isp1507.h". Am I just not finding it or is there not such a file written?

Klievan, 2021-11-23:
Hi Tijs,

Correct, there is no such file. You have to write your own communication between the BLE module and the OCTA. There are two GPIO lines available for communication between the two. Using UART is highly recommended as both the OCTA and the nRF53832 have UART drivers in place.

You can flash the nRF52832 module through the BLE debug port on the OCTA board. I suggest you use the nRF5 SDK to write your BLE application, as we did in class.

Are you able to continue using this information?

MatsDeMeyer, 2021-11-25:
There already is a somewhat of a driver that we use for our firmware over the air updates.

The FOTA logic itself is not included in this version of the stack, but the BLE UART part is.
It is surrounded by if USE_BOOTLOADER defines however, because it uses the same UART that is used on the P2 connector of the octa.

I would suggest you to change this to something like if (USE_BOOTLOADER or USE_BLE) or something similar, so that you can enable/disable this BLE uart trough defines in the Makefile.

MatsDeMeyer, 2021-11-25:
have a look at the `core/platform/octa/src/uart.c` and `platform.c` files

SamGeens, 2021-11-25:
Which PIN's are used for uart in the nRF bluetooth chip?

MatsDeMeyer, 2021-11-25:
```c
#define RX_PIN_NUMBER  26
#define TX_PIN_NUMBER  27
```

MatsDeMeyer, 2021-11-25:
snippet of BLE UART communication:

```c
/**@brief Function for handling the data from the Nordic UART Service.
 *
 * @details This function will process the data received from the Nordic UART BLE Service and send
 *          it to the UART module.
 *
 * @param[in] p_nus    Nordic UART Service structure.
 * @param[in] p_data   Data to be send to UART module.
 * @param[in] length   Length of the data.
 */
/**@snippet [Handling the data received over BLE] */
static void nus_data_handler(ble_nus_t * p_nus, uint8_t * p_data, uint16_t length)
{
    uint32_t err_code;

    NRF_LOG_DEBUG("Received data from BLE NUS. Writing data on UART.\r\n");
    NRF_LOG_HEXDUMP_DEBUG(p_data, length);

    for (uint32_t i = 0; i < length; i++)
    {
        do
        {
            err_code = app_uart_put(p_data[i]);
            if ((err_code != NRF_SUCCESS) && (err_code != NRF_ERROR_BUSY))
            {
                NRF_LOG_ERROR("Failed receiving NUS message. Error 0x%x. \r\n", err_code);
                APP_ERROR_CHECK(err_code);
            }
        } while (err_code == NRF_ERROR_BUSY);
    }
//    if (p_data[length-1] == '\r')
//    {
//        while (app_uart_put('\n') == NRF_ERROR_BUSY);
//    }

}
```

yalcinpak, 2021-12-02:
> ```c
> #define RX_PIN_NUMBER  26
> #define TX_PIN_NUMBER  27
> ```
> 
> This should be defined in `components/boards/pca10040.h`

I would like to say pin number is different for pca10056.h
#define RX_PIN_NUMBER  8
#define TX_PIN_NUMBER  6

Klievan, 2021-12-02:
Correct, PCA10056 is the devkit for the nRF52840. PCA10040 is the devkit for the nRF52832, the BLE chip on the OCTA. Therefore you should build your application using the PCA10040 build-files, so your flash and RAM is mapped correctly into the BLE chip on the OCTA.

> This should be defined in components/boards/pca10040.h
> 
> I would like to say pin number is different for pca10056.h
> #define RX_PIN_NUMBER  8
> #define TX_PIN_NUMBER  6
> ```

Indeed they are different, because those are the RX and TX pins that are mapped on the devkit by default.

yalcinpak, 2021-12-09:
#define RX_PIN_NUMBER  26
#define TX_PIN_NUMBER  27
I defined rx and tx pin in components/boards/pca10040.h
Then I uploaded the program ble_app_uart to my ble module on octa 

I activated rx callback at core/platform/octa/src/uart.c           with code below
static void (*BLE_UART_RX_Callback)(void) = 1;
Then I uploaded the program temperature-humidity example to my octa board.

I can not see any communication between ble module and octa board.
I am using mobaxterm to see output data of octa board
I am using nrfconnect app and I can open nordic_uart service. But I didn't see on rx or tx characteristic.

There are a lot of variables to understand what is wrong on my uart communication. Could you give me a tip?

MatsDeMeyer, 2021-12-09:
Sorry, missed this one earlier.

What and how are you sending over the UART lines of the NRF?

Have you enabled the BLE_UART on the octa? This is a bit messy in the current repo. It uses the same UART as on the P2 connector, and is not used. Other groups have fixed this using USER defines, I can help you with this if you want.
Also, I would recommend to do a normal (blocking) HAL_UART_Receive first on the BLE_UART before using the interrupt based approach.

yalcinpak, 2021-12-10:
I sent data over nordic uart service on app. I will check my code here.
I enabled BLE_UART but you said it uses p2 uart. Then I enabled P2 uart also(still not work). But  i think i have another problems also my project.
I will try my own service I may need some advice in future. Thank you for your help.

yalcinpak, 2021-12-27:
[image]
I have added these codes to Makefile.core, then
[image]
I added USE_BLE=1 to my makefile in example Temperature humidity , then
[image]
I add  ||USE_BLE to the line 563. But i think i have problem here. Because line 564,565,566,567,568 looks as inactive region. I couldn't make it active. What do you think about it?
then,
[image]
I initialize BLE without receiving error. I couldn't see any data on PD5 or PD6 pins.