MOSAIC-LoPoW/octa-stack-students #26: BLE power efficiency

n00bhax, 2021-11-25:
I'm from the beer glass group 🍺 and we would like some guidance in how we can make our glass nodes the most power efficient.
Right now, our tracking works via a continuous connection between a glass and all gateways, which is of course not ideal.
In terms of roles, we should probably make our glass a broadcaster and the gateways observers to retrieve RSSI values. 
How do we go about this? 
Another idea of ours was to only broadcast when the accelerometer detects movement. 
Are there any other ways to achieve power efficiency?

Klievan, 2021-11-25:
Hi n00bhax! 

Advertising on your glass nodes is very power efficient, especially if you keep the advertising rate above 300ms which is should be plenty if your gateways are continuously scanning (maybe even still too fast). Stopping the advertisements when your glasses are idle does not seem like a good idea since your module could die for some reason and you would never know it is not broadcasting anymore. You can have 2 different intervals though, one for moving (eg every 100ms) and one for when idle (1000ms). Advertising every 1000ms on an nRF52 consumes about 10-20uA continuously, which gives you months/years of operation on a single coincell. 

Below some examples which should help you:
https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v15.2.0/ble_sdk_app_beacon.html

https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v15.2.0/ble_sdk_app_proximity.html (this approach is connection based, but still very low power. Look into it!)

n00bhax, 2021-11-25:
Is there any example for an observer? I can't seem to find one

Klievan, 2021-11-25:
There isn't indeed, but you can just take a scanner example, and just remove the connecting part. Just keep the reading part of the advertisement.

https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v15.0.0/ble_sdk_app_blinky_c.html

n00bhax, 2021-11-25:
Yeah, I have been trying to do that, but I find it hard to understand all the code completely. Will you be attending the next class? That would make asking questions a little bit easier

Klievan, 2021-11-25:
Chances are I wont be making it to next class. The code is fairly simple though, below a third party developer which has published their own observer which offloads all received beacons over UART.

https://github.com/nicslabdev/NRF52840-Beacon_Scanner/blob/master/main.c

If you still are struggling after reviewing this example, we can sit together and discuss your code over a call. Feel free to contact me over Teams or e-mail to schedule.

n00bhax, 2021-11-30:
That project is using an old version of the SDK. Is it common to use older versions when newer versions are available? Or should we try to update it to the newest one?

Klievan, 2021-11-30:
I suggest updating to the newest one. Ideally morph an example from the latest SDK into this other application. I should be in class later today if you require further explanation

closed 2021-12-03 by MatsDeMeyer