MOSAIC-LoPoW/octa-stack-students #27: Fingerprinting visualization and libraries

n00bhax, 2021-11-25:
What is the best way to visualize the results of our fingerprinting?
At first, we used anvil as a front end web interface, but it's starting to ask for money so we need to find an alternative. 
Grafana and ThingsBoard look promising. Which one would you suggest?
Are there any offline front ends that work out of the box? 

It would also be handy to find a library for fingerprinting for our back-end to help with the database creation and finding the closest match. Do these exist?

michielaernouts, 2021-11-25:
We have seen student demos with ThingsBoard to show fingerprinting results in past years, so we know that will work.
You could also write your own python script with a visualization framework that show your results in a web browser, there are some solutions out there (example: https://plotly.com/dash/).

n00bhax, 2021-11-25:
And how would you tackle the database? Currently, we're thinking of using SQL

michielaernouts, 2021-11-25:
You could also just store your training database as a numpy array, I think this will work faster than always querying an SQL database. I can brainstorm with you on this on Tuesday :slightly_smiling_face:

n00bhax, 2021-11-25:
That's actually how we are doing it now, but we just use a for loop to go over all entries in the database. We thought that maybe SQL could find close matches more efficiently. Brainstorming on Tuesday sounds good!

closed 2021-12-03 by MatsDeMeyer