MOSAIC-LoPoW/octa-stack-students #29: uplink payload formatter decode problem

yalcinpak, 2021-11-25:
I receive same data. My desired temperature value is about 17.78 (as a float )

[image]
my uplink payloadFormatter is below

function Decoder(bytes, port) {
  var decoded = {};
  decoded.temperature = ((bytes[0] << 8) | bytes[1]) / 100.00;
  return decoded;
}

michielaernouts, 2021-11-25:
Are you sure you are calling the correct bytes (bytes[0] and bytes[1]? There is a 4 byte preamble (20 40 00 04) added by the octa-stack, that is not your data.

MatsDeMeyer, 2021-11-29:
The first 4 bytes is indeed a preamble, with the last byte being the size of your actual payload (4).
You can discard those first bytes, and only look at the last four.

I have not worked with the payloadformatter before, but it indeed seems you are only using two bytes (of the preamble) instead of your 4 byte temperature data.

In my opinion, the payload formatter is a bit of an unnecessary intermediate step.
I would focus on getting the raw data in your backend/server using a python script and properly parsing it there.

MatsDeMeyer, 2021-11-29:
By quickly throwing your last 4 bytes in an online converter, i get 17.6672764 as the float value so your device side looks correct.

This converter works really well for decoding hexstrings, for your float you need the **Float - Little Endian**:
https://www.scadacore.com/tools/programming-calculators/online-hex-converter/

MatsDeMeyer, 2021-11-30:
Here is a the documentation about the MQTT api from TTN:
https://www.thethingsindustries.com/docs/integrations/pubsub/mqtt-client/
The goal is to do these step in a python script on your server, there should be plenty of examples on how to use MQTT in python.

closed 2021-12-03 by MatsDeMeyer