MOSAIC-LoPoW/octa-stack-students #32: Not possible to read gps and light sensor data

GBlankers, 2021-11-30:
We tried to merge our working gps code and light sensor code but only one of the two is working depending on which is initialized last. To debug a gps register is printed:
[image]

Light sensor initialized last:
[image]

[image]

GPS initialized last:

[image]

[image]

MatsDeMeyer, 2021-11-30:
Mistake on our side.

Both the i2cHandle variable of the TCS3472 & GPS driver are declared in their respective header files, both with the same name:
```c
I2C_HandleTypeDef *hi2cLib
```

Thus, when initializing the tcs sensor first for example, this variable gets set to common_I2C.
Whe calling Firefly_initialize() after that, the **same** variable gets set to the i2c handle of whatever connecter is defined in the Makefile.

Renaming one or ideally both of them to a more unique name, e.g. hi2cLib_tcs & hi2cLib_gps will fix this.

MatsDeMeyer, 2021-12-01:
or even better, removing these from the header file

closed 2021-12-03 by MatsDeMeyer