MOSAIC-LoPoW/octa-stack-students #33: UUID ble

n00bhax, 2021-11-30:
ble_advdata_uuid_find verwacht uuid in een raar vorm. 
halp
[image]

MatsDeMeyer, 2021-12-01:
You need to be a bit more specific here, I don't really see the issue or question from your screenshot.

SamGeens, 2021-12-01:
The `ble_advdata_uuid_find` funtion matches the uuid given as parameter with the discovered data. We are trying to give this uuid as parameter but we have troubles with the uuid syntax. We already discovered that the uuid is a [struct](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.s132.api.v2.0.0%2Fstructble__uuid__t.html) and that the uuid expects 16-bit UUID value or octets 12-13 of 128-bit UUID..

Klievan, 2021-12-01:
If you are not registering your UUID using sd_ble_uuid_vs_add() you might have to set your UUID type to UNKNOWN, not 100% sure. Apart from that, it should be just as the API says, combining octet 12 and 13 into a uint16_t. 

In a BIG ENDIAN UUID, those octets are marked here by an xx.
ABCDxxxx-EF12-3456-7890-ABCDEF123456

Note that the stack works in little endian, so your octets will be reversed, so if I'm not mistaken that should be 0xCD and 0xDE. So that becomes 0xCDDE, or the other way around. I'll let you figure that out on your own ;)

n00bhax, 2021-12-02:
We tried everything you suggested, but we still can't filter on our beacon uuid.
Changing the .type to UNKNOWN didn't work. 
The next step was to register the uuid. Did we do this correctly? 

```
/**@brief 128-bit UUID base List. */
static const ble_uuid128_t m_base_uuid128 =
{
   {
       0x01, 0x12, 0x23, 0x34, 0x45, 0x56, 0x67, 0x78,
       0x89, 0x9a, 0xab, 0xbc, 0xcd, 0xde, 0xef, 0xf0
   }
};
```

In the main:

```
// add UUID to the BLE stack's table
    uint32_t   err_code;
    ble_uuid_t service_uuid;

    err_code = sd_ble_uuid_vs_add(&m_base_uuid128, &service_uuid.type);
    if (err_code != NRF_SUCCESS){
        NRF_LOG_ERROR("iets fout met uuid add: %d", err_code);
    }
```

In the ble_evt_handler
```
 case BLE_GAP_EVT_ADV_REPORT:
        {
            const ble_uuid_t uuid = {.type = 0x02, .uuid = 0xCDDE};
            if(ble_advdata_uuid_find(m_scan.scan_buffer.p_data, m_scan.scan_buffer.len, &uuid)) {
                NRF_LOG_RAW_HEXDUMP_INFO (m_scan.scan_buffer.p_data, m_scan.scan_buffer.len);
                NRF_LOG_RAW_INFO ("----------------------------------\r\n");

                //sd_ble_gap_rssi_get()
            }
        }
```

You can also have a look at our git for the full code: 
https://github.com/SamGeens/IoT_Glass_tracker/blob/beacon_scanner/BLE/nRF5_SDK_17.1.0_ddde560/projects/gateway/beacon_scanner/main.c

SamGeens, 2021-12-02:
@Klievan we are still having issues with the UUID. After our recent changes we can’t find the beacon anymore (both in nRF app and custom beacon scanner). Could you help us?

Klievan, 2021-12-03:
One of your APP_ERROR_CHECK's() is failing. Add the debug flag in your makefile to see where your run-time error is. In your CFLAGS add
`CFLAGS += -DDEBUG`
In your ASMFLAGS add
`ASMFLAGS += -DDEBUG`

Most likely you will see something like "OUT OF MEMORY ERROR om main.c:123" which would mean that error occurred in your main.c file on line 123. Google is your friend here, 90% of the time there is someone who has exactly the same error. You can also just look up the documentation of that function and it will also accurately describe why you are getting that error.

NaderMotee, 2021-12-03:
[image]

Klievan, 2021-12-03:
All right, now go and look what you are checking for on that line. Go look up the function that is not returning success. In the API docs it will tell you what those errors mean for each function.

SamGeens, 2021-12-04:
@Klievan 

After searching for a while and trying a few options I'm still stuck with this one. I found that the NRF_ERROR_INVALID_PARAM (above error) is because of these 2 lines:

```
advdata.uuids_complete.p_uuids = m_adv_uuids;
advdata.uuids_complete.uuid_cnt = sizeof(m_adv_uuids) / sizeof(m_adv_uuids[0]);
```

Configuration of `m_adv_uuids`: 
`static ble_uuid_t m_adv_uuids[] = {{BLE_UUID_DEVICE_INFORMATION_SERVICE, BLE_UUID_TYPE_VENDOR_BEGIN}};  /**< Universally unique service identifiers. */`

When commenting those 2 lines and changing `NRF_SDH_BLE_VS_UUID_COUNT` to 1 in the config file the following error came accros in `sd_ble_uuid_vs_add` this was fixable with actually following the RAM location advise. 
[image]

After doing this the beacon is advertising again but Manufacturing data and UUID is empty.

Klievan, 2021-12-04:
Correct, because you commented out the lines that advertise your UUIDs. I took a quick look at your code, in your code you have
`static ble_uuid_t m_adv_uuids[] = {{BLE_UUID_DEVICE_INFORMATION_SERVICE, BLE_UUID_TYPE_VENDOR_BEGIN}};`
That doesn't look 100% right, you're trying to advertise a 16-bit service as a 128-bit service. Try replacing BLE_UUID_DEVICE_INFORMATION_SERVICE with APP_BEACON_UUID

Klievan, 2021-12-04:
Also, try to think more logical when debugging before asking for help. For example, in this case you could've thought like this:
1. The check on line 214 is failing. This seems to be the error_code returned by the function `ble_advdata_encode`. 
2.[ The API documentation](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v15.0.0%2Fgroup__ble__sdk__lib__advdata.html) says that the parameter p_data is structured wrongly, but I could have determined that also from the name of the error (NRF_ERROR_INVALID_PARAM).
3. `p_data` was fine before I edited the manufacturer data and adding `advdata.uuids_complete`, so most likely there's something wrong there
4. If I comment out the `advdata.uuids_complete` assignments, I do not get the error anymore so this must be the part that is invalid. 
5. The UUID-count resolves back to 1, which is correct so it must be my `m_adv_uuids`
6. < Here comes the part where you notice that you're not actually using the right UUID > You try to replace it.
7. If it works, yay. If it does not, look up the documentation on how to actually structure m_adv_uuids etc.

SamGeens, 2021-12-08:
Thank you for your help and the extra step guide. We reached step 5-6 before asking for help. At this point we were stuck and spent more than 5 hours (group effort) on this, so we thought this was the appropriate time to ask a question.

Your aswer indeed pushed us into the right direction and we were able to resolve this issue yesterday. Thanks again for the help, enjoy your holiday!

closed 2021-12-08 by n00bhax