MOSAIC-LoPoW/octa-stack-students #34: Power profiler connection

yannisarnouts, 2021-12-02:
Hi

How can I connect the power profiler with the Octa?

Thanks!

Klievan, 2021-12-02:
There's two ways, one which uses an nRF52 devkit and one which does not but requires a JLINK programmer.
https://infocenter.nordicsemi.com/topic/ug_ppk/UG/ppk/PPK_user_guide_PPK_on_customHW_with_nRF5xDK.html
https://infocenter.nordicsemi.com/topic/ug_ppk/UG/ppk/PPK_user_guide_PPK_on_customHW_standalone.html

I THINK you can just connect to the OCTA through VDD/VIN and GND, no @MatsDeMeyer ?

MatsDeMeyer, 2021-12-02:
Yes, correct! You can supply 3.3V directly to the octa.

yannisarnouts, 2021-12-02:
[image]

Okay so just like this? It either disconnects the whole time when the PPK is opened.
Or I get this error message.
[image]

yannisarnouts, 2021-12-02:
So just like this?
[image]
It connects, but every time this happens and connection is lost
[image]

Klievan, 2021-12-02:
Hi Yannis. Try reviewing your setup. I don't see a devkit beneath your powerprofiler, or a JLINK-debugger attached. Your options:
1. Place your power profiler on your BLE devkit, and switch the USB from your profiler onto the main devkit USB-port. 
2. Use a JLINK-debugger and connect it to your powerprofiling kit.
Make sure you follow the instructions on either two URL's I've sent you. It explains on how to set the switches, and where to connect your USB.

yannisarnouts, 2021-12-02:
Yes the devkit is beneath the power profiler. Not really clear in the photo indeed ☺️

Klievan, 2021-12-02:
Allright my bad! Have you connected the USB to the USB of the devkit?

yannisarnouts, 2021-12-02:
Devkit

Klievan, 2021-12-02:
What happens if you disconnect the jumpers from the PPK and try to measure "nothing"?

yannisarnouts, 2021-12-02:
Okay so DUT select is set to External and power select to Reg.

First time: it starts but I get an error, because there is no connection.

[image]

I try to connect again, now it works and measures
[image]

Klievan, 2021-12-02:
While you are measuring, try connecting the device. Does the power profiling kit disconnect when you attach the jumper cables or do you get a normal reading?

MatsDeMeyer, 2021-12-02:
I'm afraid the startup current is too high in that case. What happens if you take of the GPS and murata module?

yannisarnouts, 2021-12-02:
Yes it works now, afterwards I can reconnect the modules.
Thanks @Klievan and @MatsDeMeyer 🥳 🥇

MatsDeMeyer, 2021-12-02:
You can also neutralize this startup peak by connecting a battery, and disconnecting this after a few seconds.