MOSAIC-LoPoW/octa-stack-students #35: Buzzer struggle

thibeaudemol, 2021-12-02:
I tried to follow a youtube tutorial which was using cubeMX (https://www.youtube.com/watch?v=AY8t7lsiG7I) to be able to make a PWM signal and drive the buzzer.

[image]
(picture of cubeMX)

Sadly, it didn't work for me although i followed all his steps. 
In my head there are 2 options where the fault can be.

-He is using a different type of buzzer with GND,VCC and input and i'm using a buzzer (pkm22epp-40) with only 2 inputs (PWM signal and GND i assume) 

-I connected the wrong pins (PWM signal is generated on pin A0 i think) 

[image]
[image]
(setup)

Klievan, 2021-12-02:
First thing that comes to mind is that connecting jumper-cables like that sometimes causes continuity issues. Do you happen to have a multimeter that you can use to check for that?

thibeaudemol, 2021-12-02:
No, tuesday i asked Mats if i could take one home but he said it was not possible :/

MatsDeMeyer, 2021-12-02:
Something else, did you start your timer after configuring it?

Can you share your timer init and `HAL_TIM_MspPostInit()` functions here?

thibeaudemol, 2021-12-02:
timer init (generated using cubeMX)

static void MX_TIM2_Init(void)
{

  /* USER CODE BEGIN TIM2_Init 0 */

  /* USER CODE END TIM2_Init 0 */

  TIM_ClockConfigTypeDef sClockSourceConfig = {0};
  TIM_MasterConfigTypeDef sMasterConfig = {0};
  TIM_OC_InitTypeDef sConfigOC = {0};

  /* USER CODE BEGIN TIM2_Init 1 */

  /* USER CODE END TIM2_Init 1 */
  htim2.Instance = TIM2;
  htim2.Init.Prescaler = 692-1;
  htim2.Init.CounterMode = TIM_COUNTERMODE_UP;
  htim2.Init.Period = 255-1;
  htim2.Init.ClockDivision = TIM_CLOCKDIVISION_DIV1;
  htim2.Init.AutoReloadPreload = TIM_AUTORELOAD_PRELOAD_DISABLE;
  if (HAL_TIM_Base_Init(&htim2) != HAL_OK)
  {
    Error_Handler();
  }
  sClockSourceConfig.ClockSource = TIM_CLOCKSOURCE_INTERNAL;
  if (HAL_TIM_ConfigClockSource(&htim2, &sClockSourceConfig) != HAL_OK)
  {
    Error_Handler();
  }
  if (HAL_TIM_PWM_Init(&htim2) != HAL_OK)
  {
    Error_Handler();
  }
  sMasterConfig.MasterOutputTrigger = TIM_TRGO_RESET;
  sMasterConfig.MasterSlaveMode = TIM_MASTERSLAVEMODE_DISABLE;
  if (HAL_TIMEx_MasterConfigSynchronization(&htim2, &sMasterConfig) != HAL_OK)
  {
    Error_Handler();
  }
  sConfigOC.OCMode = TIM_OCMODE_PWM1;
  sConfigOC.Pulse = 0;
  sConfigOC.OCPolarity = TIM_OCPOLARITY_HIGH;
  sConfigOC.OCFastMode = TIM_OCFAST_DISABLE;
  if (HAL_TIM_PWM_ConfigChannel(&htim2, &sConfigOC, TIM_CHANNEL_1) != HAL_OK)
  {
    Error_Handler();
  }
  /* USER CODE BEGIN TIM2_Init 2 */

  /* USER CODE END TIM2_Init 2 */
  HAL_TIM_MspPostInit(&htim2);

}

thibeaudemol, 2021-12-02:
HAL_TIM_MspPostInit: also generated by cubeMX

void HAL_TIM_MspPostInit(TIM_HandleTypeDef* htim)
{
  GPIO_InitTypeDef GPIO_InitStruct = {0};
  if(htim->Instance==TIM2)
  {
  /* USER CODE BEGIN TIM2_MspPostInit 0 */

  /* USER CODE END TIM2_MspPostInit 0 */

    __HAL_RCC_GPIOA_CLK_ENABLE();
    /**TIM2 GPIO Configuration
    PA0     ------> TIM2_CH1
    */
    GPIO_InitStruct.Pin = GPIO_PIN_0;
    GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
    GPIO_InitStruct.Pull = GPIO_NOPULL;
    GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_LOW;
    GPIO_InitStruct.Alternate = GPIO_AF1_TIM2;
    HAL_GPIO_Init(GPIOA, &GPIO_InitStruct);

  /* USER CODE BEGIN TIM2_MspPostInit 1 */

  /* USER CODE END TIM2_MspPostInit 1 */
  }

}

thibeaudemol, 2021-12-02:
did you start your timer after configuring it --> YES HAL_TIM_PWM_Start(&htim2,TIM_CHANNEL_1);

MatsDeMeyer, 2021-12-02:
At first glance, this looks fine to me.
How do you now trigger the buzzer?

thibeaudemol, 2021-12-02:
I put this code in a different project, it should just always be buzzing

MatsDeMeyer, 2021-12-02:
Does it work in the different project?

We use PWM for the RGB led on the octa, and to write an RGB value to it using PWM we do this:

```c
void OCTA_LED_PWM_Init(void)
{
    MX_TIM3_Init();
    MX_TIM4_Init();
    /* USER CODE BEGIN 2 */

    HAL_TIM_PWM_Start(&htim3, TIM_CHANNEL_3);
    HAL_TIM_PWM_Start(&htim4, TIM_CHANNEL_3);
    HAL_TIM_PWM_Start(&htim4, TIM_CHANNEL_4);
    OCTA_LED_set_rgb(0, 0, 0);
}

void OCTA_LED_set_rgb(uint8_t red, uint8_t green, uint8_t blue)
{
  htim4.Instance->CCR3 = 65535 - red*256;
  htim3.Instance->CCR3 = 65535 - green*256;
  htim4.Instance->CCR4 = 65535 - blue*256;
}
```

thibeaudemol, 2021-12-02:
[image]

I do exactly the same here

Klievan, 2021-12-02:
Do you happen to have a loose LED laying around which you can probe? Not sure what kind of resistor you have there?

thibeaudemol, 2021-12-02:
Smart idea! I do have a led i think, and that's a 100 ohm resistor.

thibeaudemol, 2021-12-02:
When i use a LED, it turns on but doesn't change in light intensity

thibeaudemol, 2021-12-02:
[image]

Using this code, it just remains on the whole time

MatsDeMeyer, 2021-12-02:
Is this with values between 0 & 255 or values between 0 & uint32_t MAX?

MatsDeMeyer, 2021-12-02:
can you push this code to a branch of your repository?

thibeaudemol, 2021-12-02:
I put the code in the issues tab of our project, didn't want to mess up the branching. This is the different project (for testing purpouses) with code generated by cubeMX (same as the youtube video)

MatsDeMeyer, 2021-12-02:
Don't see any fundamental mistakes, only difference I see is the prescaler/period values.
I use:
```c
/* USER CODE END TIM3_Init 1 */
  htim3.Instance = TIM3;
  htim3.Init.Prescaler = 0;
  htim3.Init.CounterMode = TIM_COUNTERMODE_UP;
  htim3.Init.Period = 65535;
  htim3.Init.ClockDivision = TIM_CLOCKDIVISION_DIV1;
  htim3.Init.AutoReloadPreload = TIM_AUTORELOAD_PRELOAD_DISABLE;
```

And then as stated before, set the values as follows (with green being an uint8_t value):
```c
htim3.Instance->CCR3 = 65535 - green*256;
```

I would put the prescalers like this and change your main loop to
```c
while (1)
  {
    while(value<255){
      htim2.Instance->65535 - value*256; // vary duty cycle
      value+=20;
      HAL_Delay(500);
      
    }
    value=0;
  }
```

MatsDeMeyer, 2021-12-02:
If that doesn't work, I can help you on tuesday afternoon

thibeaudemol, 2021-12-02:
Didn't work :'(

I'll stop with the buzzer for today and start with the servo

MatsDeMeyer, 2021-12-09:
Do you know what the fix was? (in your CubeMX project)

thibeaudemol, 2021-12-09:
Fixed it by stealing pwm from leds

closed 2021-12-09 by thibeaudemol