MOSAIC-LoPoW/octa-stack-students #39: Debugging on the nRF in VSCode (BLE)

n00bhax, 2021-12-02:
How do you use the cortex-debugger with the nRF boards?
We tried multiple launch.json configurations. 
One of them started the debugger seemingly successful, but when we add a breakpoint, the program stopped.

[image]

n00bhax, 2021-12-02:
Apparently it's our code that's causing the problems

n00bhax, 2021-12-02:
It looks like the debugger is causing our program to crash, while it runs normally without it.

SamGeens, 2021-12-02:
It seems that crashing after continouing from a debug point is the 'expected' behaviour when using a soft device:
https://devzone.nordicsemi.com/f/nordic-q-a/32973/nrf_breakpoint_cond-in-debug

Klievan, 2021-12-03:
Indeed. So you can't break once you've got the softdevice initialized. The reason for this is that BLE needs 100% accurate timing to sync its communication. When you stop at a breakpoint your hardware timers are still going to be running, so when you resume and the clock suddenly jumped by a couple of seconds it's going to go into a hardfault as it thinks the crystal is broken. You can still use breakpoints just to see if you're reaching a specific line, but after that you've got to reset the device. This is what I usually do to debug BLE issues.

closed 2021-12-05 by n00bhax