MOSAIC-LoPoW/octa-stack-students #40: nRF52840 dev kit - arduino leonardo uart connection

yalcinpak, 2021-12-02:
I struggled with uart. I couldn't make successful connection. Then I decided to make problem narrow perspective. I wrote some code to my arduino that led will open if receive 1 on uart(RX). I wrote some code to my nrf board that nrf will write same data to uart(TX) which received from bluetooth app. My arduino code is working correctly if I send data from computer via usb cable.  My nrf code is working correctly with computer usb as well. Unfortunatelly these two boards can not work together now. I adjusted baud rate to 115200 for 2 boards.
note( nrf can't work properly if I use arduino usb to supply when sending data)

Nrf 5V          <---->   arduino 5v
Nrf gnd        <---->   arduino gnd
Nrf P0.06(tx) <---->   arduino 0 (rx
[image]
)

Klievan, 2021-12-03:
Really good that you're taking one factor out of the equation. Sadly, I'm afraid that the Arduino will run it's UART on 5V. Normally the nRF52840 devkit SHOULD be protected against 5V blasting into it's GPIO ports but the 3V from the nRF52840 might not be enough to trigger UART communication on the Arduino.

What is your final plan with BLE? Are you going to use the onboard OCTA ble chip, or are you going to use an nRF52 devkit?

yalcinpak, 2021-12-05:
I hope i didn't damage the boards. We would like to use OCTA ble chip if I overcome. I just tired about using different flasher program but i will manage. I thank you for your informative message.

closed 2021-12-07 by yalcinpak