MOSAIC-LoPoW/octa-stack-students #42: BLE not working on OCTA

TomRaeves, 2021-12-07:
We can't run our BLE program on the octa stack but it works on the NRF board (10056). We tried running blinky on both and this works on the octa and on the NRF board.

We added extra lines to our project makefile for 10040 which we also had to add in the 10056.

Klievan, 2021-12-07:
If you have RTT logging enabled you can connect a debugger to the onboard nRF and read it out through an RTT client or viewer

TomRaeves, 2021-12-07:
When we do the RTT logging nothing comes up in the window although we enabled RTT logging. Is it possible that you could help us on thursday ?

Klievan, 2021-12-09:
Have you tried erasing the OCTA nRF and reflashing the softDevice? The nRF on the octa needs SoftDevice S132

TomRaeves, 2021-12-09:
Is this just executing the command "make flash_softdevice" because we already tried this ?

MatsDeMeyer, 2021-12-09:
Try erasing it first as Ivan said.

https://infocenter.nordicsemi.com/index.jsp?topic=%2Fug_gsg_keil%2FUG%2Fgsg%2Ferase_board.html

TomRaeves, 2021-12-09:
This does not work either.

michielaernouts, 2021-12-09:
We can solve issues in the 'this code is not working' category by sitting with you and debugging your code step by step. Then again, that is not up to us... We can assist you in finding relevant reference material if there is a more detailed bug report with specific issues.

MatsDeMeyer, 2021-12-09:
If the blinky example does work but you program does not, it really has to have something to do with:
1) your code
2) the config/makefile of your custom program
3) the way you are flashing your custom applications to the octa

Please try to go step by step over this again, starting from the most simple application on the octa first, maybe start from blinky and add pieces to until it breaks again.

Again trying to get RTT logging to work again will also help you massively, as you will be able to see what actually goes wrong.

TomRaeves, 2021-12-09:
it has been solved in the meantime. We just cloned the project from the guide and now the makefiles and such do work

closed 2021-12-16 by MatsDeMeyer