MOSAIC-LoPoW/octa-stack-students #43: RTC alarm

MatthiasW1004, 2021-12-07:
I am trying to make the RTC alarm interrupt work, but the interrupt doesn't get triggered when it reaches the correct time. My program also gets stopped by the debugger but I have no clue why the debugger issues a reset or a halt.

[image]

MatsDeMeyer, 2021-12-07:
Where does the debugger stop? Do you have a breakpoint somewhere?

MatthiasW1004, 2021-12-07:
If I try to open the file, it says that it isnt there.
[image]

MatthiasW1004, 2021-12-07:
I can find the file in my windows explorer at that location, but it does not show in vs code.

MatsDeMeyer, 2021-12-08:
Can you share how you set up the interrupt?

MatthiasW1004, 2021-12-09:
I just enabled the RTC and alarm settings in CubeMX and copied the code. But it never uses the HAL_NVIC_EnableIRQ(EXTI18_IRQn); command. (RTC alarm interrupt through EXTI line 18)

MatthiasW1004, 2021-12-09:
[image]

MatsDeMeyer, 2021-12-09:
You should enable the IRQ (the line that is commented out), which priority do you give this interrupt?

MatsDeMeyer, 2021-12-09:
Also have a look at the rtc.c file of `core/platform/octa/src`, as we already initialize the RTC there.
Copy pasting your MX_RTC_INIT function somewhere and calling this afterwards can cause issues

MatthiasW1004, 2021-12-09:
I just deleted the MX_RTC_INIT function and used the RTC_INIT from rtc.c. I also added a function to set the alarm. I still have the same error as before when I try to run it. 
[image]

MatthiasW1004, 2021-12-09:
Can you maybe join our teams channel for a moment?

MatthiasW1004, 2021-12-09:
[image]
[image]
This is my code I try to run, I also added __HAL_RTC_ALARM_EXTI_ENABLE_IT(); in the RTC_Init function

MatsDeMeyer, 2021-12-09:
The fact that you are getting an error handler might be because you are calling RTC_Init(), while this is already done in the Initialize_Platform function. Same with HAL_Init and Systemclock_config.

I would suggest to place a breakpoint at the error_handler and looking at the call stack in your debug session.

MatthiasW1004, 2021-12-16:
Do you have a code example on how to write to rtc configuration during runtime? I have been struggling with this for 3 weeks now and cant find a solution.

MatsDeMeyer, 2021-12-16:
We update the datetime at runtime with `void RTC_Set_Time_From_Unix(uint32_t UnixNum)` in `/core/platform/octa/src/rtc.c`.

What exactly goes wrong?

MatthiasW1004, 2021-12-16:
We might have fixed it, we want to set an alarm in the app. But it keeps telling the RTC is locked.

MatsDeMeyer, 2021-12-16:
https://www.st.com/resource/en/application_note/an3371-using-the-hardware-realtime-clock-rtc-in-stm32-f0-f2-f3-f4-and-l1-series-of-mcus-stmicroelectronics.pdf here is the application note of STM

MatthiasW1004, 2021-12-16:
Thank you, I can edit the time and alarm now. I still get an error from the interrupt that it gets stuck in an infinite loop. I looked it up and I found the solution but it doesnt work for me. Following link summarizes problem and solution: https://community.st.com/s/question/0D53W00000BsF0FSAV/program-stops-in-wwdgirqhandler-infinite-loop-after-enabling-adc-interrupts

MatsDeMeyer, 2021-12-16:
Do you use the correct IRQ handler for alarms? WWDG IRQ handler does indeed seem like it does not find the correct IRQ handler and just takes a random one, as the WWDG is not enabled.

MatthiasW1004, 2021-12-16:
I do think so?
[image]
[image]

MatsDeMeyer, 2021-12-16:
The IRQHandler is commented out in your screenshot, you only have your callback

MatthiasW1004, 2021-12-16:
I know, I did that because I get an error if I define the handler myself.