MOSAIC-LoPoW/octa-stack-students #46: BLE Advertising size to small

SamGeens, 2021-12-09:
Hi, 

When advertising our custom UUID we get the following error
[image]

Line 214 is the following line: 
```
err_code = ble_advdata_encode(&advdata, m_adv_data.adv_data.p_data, &m_adv_data.adv_data.len);
APP_ERROR_CHECK(err_code);
```

As stated in the NRF documentation this means the pointer buffer is to small.
[image]

The actual buffer struct is now:
```
/**@brief Struct that contains pointers to the encoded advertising data. */
static ble_gap_adv_data_t m_adv_data =
{
    .adv_data =
    {
        .p_data = m_enc_advdata,
        .len    = BLE_GAP_ADV_SET_DATA_SIZE_MAX
    },
    .scan_rsp_data =
    {
        .p_data = NULL,
        .len    = 0

    }
};
```

`BLE_GAP_ADV_SET_DATA_SIZE_EXTENDED_MAX_SUPPORTED` is defined as:
```
/**@defgroup BLE_GAP_ADV_SET_DATA_SIZES Advertising data sizes.
 * @{ */
#define BLE_GAP_ADV_SET_DATA_SIZE_MAX                    (31)   /**< Maximum data length for an advertising set.
                                                                     If more advertising data is required, use extended advertising instead. */
/**@}. */
```

According to this the fix should be using the extended advertising size. I found the size of this [here](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v15.0.0%2Fgroup__ble__sdk__lib__advdata.html). But when I use the error is gone but nothing is displayed in the serial communication log and the UUID is not being advertised.

Klievan, 2021-12-09:
How many bytes are you trying to fit into your advertising data? You can use extended advertising but this needs to be enabled at both the scanner and advertiser. Take a look at the RSCS(Running speed & cadence service) example which uses extended advertising to fit enough data in their advertising. They both have a peripheral and central example.

Klievan, 2021-12-09:
Also, removing other data you don't need from your advertising struct will free up more space you can use, but as you see 31 bytes is the max for regular advertising.

SamGeens, 2021-12-09:
> Also, removing other data you don't need from your advertising struct will free up more space you can use, but as you see 31 bytes is the max for regular advertising.

This was indeed the case, our app was still advertising some data even though this was not listed the nrf app.

Removing the `flags`, 'name_type' and 'APP_BEACON_INFO_LENGTH' freed up space for the UUID.
```
    //manuf_specific_data.data.p_data = (uint8_t *) m_beacon_info;
    manuf_specific_data.data.size   = APP_BEACON_INFO_LENGTH;

    // Build and set advertising data.
    memset(&advdata, 0, sizeof(advdata));

    advdata.name_type             = BLE_ADVDATA_NO_NAME;
    advdata.flags                       = flags;
```

Thanks!

closed 2021-12-09 by SamGeens

Klievan, 2021-12-09:
Hi @SamGeens. Nice to hear! Note that the flag-field should only be omitted in two cases (as per the BLE spec):

- Either all your flag bits are 0 
- Or you do not need your advertiser to be connectible.

SamGeens, 2021-12-09:
@Klievan Thanks for the extra info! Our advertiser does not need to be connectible, so it was safe to remove.