MOSAIC-LoPoW/octa-stack-students #48: Thingy compatible code

SamGeens, 2021-12-09:
We wrote some custom code which is working on the development kit with pca10056/s140 and on the octa with pca10040/s132. Now we would like to flash this code to our thingies. But a Thingy seems kinda out of date, with a [deprecated thingy sdk](https://github.com/NordicSemiconductor/Nordic-Thingy52-FW) and with sdk13. 

The thingy should be compatible with softdevice s132 [(source)](https://nordicsemiconductor.github.io/Nordic-Thingy52-FW/) and should use board pca20020. This board file is available in the new SDK17 under components/boards. I created a pca20020/s132 directory in the project and flashing seems to be succesfull. But the led keeps flashing green, indicating that the thingy is crashing.

Custom created pca20020/s132 directory:
[image]

Any advice on this?

Klievan, 2021-12-14:
Hi Sam! The ThingySDK is indeed a bit old. Have you tried just using the regular SDK you've been using? The Thingy's run the nRF52832 so you can use those examples as well. You can copy the accelerometer driver over, should be a fairly simple driver.

SamGeens, 2021-12-14:
Falsing the PCA10040 with softdevice s132 works but I had to completely erase the thingy first with the following command:
` nrfjprog --family nRF52 --eraseall`

The standard Blinky examples is running succesfully but out custom code not, it must be crashing somewhere but we can't find out where with the Thingy. I've enabled RTT logging again but I do not see any RTT logs.

SamGeens, 2021-12-14:
Error is fixed now, it was something in our code.

closed 2021-12-14 by SamGeens