MOSAIC-LoPoW/octa-stack-students #49: Question about BLE mesh network

SamGeens, 2021-12-12:
In our glass project we have multiple glasses all broadcasting to multiple gateways (4-6). Each glass has a unique UUID, and the gateway already filters this UUID together with the RSSI of the glass. So each gateway currently receives the RSSI of each glass. 

At this point we are planning to send all the gathered data, by the gateway's, with BLE to 1 main gateway. This main gateway should send this data with LoRa to our backend.

**This means a gateway should:**

- Be both an observer and broadcaster
- Send data to main gateway when the RSSI of a glass is received

**Possible options could be:**

- Using the [nRF5 SDK for Mesh](https://www.nordicsemi.com/Products/Development-software/nRF5-SDK-for-Mesh)
Different from the current SDK, is it an option/good to mix those SDK's?
- Broadcast message after receiving one. 
The downside of this might be that broadcasting just once will not be picked up by the observer so broadcasting should be for example one minute long after receiving a message.
- Working with connections between gateways instead of connectionless. 
Less favourable option because this will be more power consuming.

Another question is the choice for LoRa. We can/will gather the data at the main gateway for x period of time before sending. Instead of sending each message directly. But even then we'll most likely be sending more than 60 times an hour. Preferably we would like to know in our backend were our glasses are with a delay of less than 20 seconds. Doing the math this means 180 LoRa messages each hour, if not more. This makes us question the use of LoRa over other technologies, like wifi.

Any ideas, opinions, advice, ... on these aspects? If easier to discuss this face-to-face, we can always schedule a teams meeting or even discuss this on the limited class hours on tuesday.

MatsDeMeyer, 2021-12-13:
If you need a delay of less than 20 seconds, LoRaWAN does not make much sense indeed. 
The Things Network is fairly strict in regards to duty-cycle/fair access policy:
https://www.thethingsnetwork.org/docs/lorawan/duty-cycle/

Is it possible to do some preprocessing on the octa? e.g. only send updates when location changes?
We will discuss this on tuesday.

SamGeens, 2021-12-13:
We also thought about preprocessing on the octa. But we thought that this wouldn't make any big difference. Our particle filter should run in our backend so the backend will need enough information about the glasses to locate them. 

Currently, we are working on the accelerometer in combination with broadcasting. But even then the number of messages won't drastically decrease. e.g. Broadcasting on glass movement (gateways receive RSSI). Let's assume we have 100 glasses in our bar, then every time a glass moves the RSSI should be sent to the backend to locate it. If someone is moving the glass (drinking/stealing) then the glass will broadcast to receive the RSSI at the gateways. This information should be sent a.s.a.p to the backend to locate the glass with the particle filter. 

A further discussion on Tuesday is perfect!

Klievan, 2021-12-13:
Hi guys! RE your BLE question

Using the nRF5 SDK for Mesh
_Different from the current SDK, is it an option/good to mix those SDK's?_
**It is possible but I strongly advise against this. Mixing SDK's is a bit difficult if you have not worked with BLE Mesh before, also for this application it does seem overkill.**
Broadcast message after receiving one.
_The downside of this might be that broadcasting just once will not be picked up by the observer so broadcasting should be for example one minute long after receiving a message._
**This is an unreliable method as since you're "echoing" advertisements from your glasses. Say gateway A B and C receive the same advertisement from a glass, they might all try to send this message at the same time which increases the chance of packet collisions drastically.**
Working with connections between gateways instead of connectionless.
_Less favourable option because this will be more power consuming._
**For me this seems to be the way to go. The nRF52832 should be able to support 5 open connections without any issue. It doesn't necessarily have to be high power as long as you tweak your connection parameters right. On top of that, one can argue that your gateways will be powered anyways. Just note that you can't be scanning AND sending at the same time, so keep this in mind when picking your connection parameters. **

Klievan, 2021-12-14:
I'm assuming you mean the Nordic UART service?

SamGeens, 2021-12-14:
Yes

Klievan, 2021-12-14:
That's an option. You can also create your own characteristic and send notifications to the "master" gateway in your own format.

closed 2021-12-14 by SamGeens