MOSAIC-LoPoW/octa-stack-students #51: BLE cant find ble_nus_data_send

TomRaeves, 2021-12-14:


TomRaeves, 2021-12-14:
[image]
[image]
[image]

TomRaeves, 2021-12-14:
yes this is the error we are getting after solving the uart_enable

Klievan, 2021-12-14:
Right, I feel like this is a similar issue no? Did you set BLE_NUS_ENABLED to 1 in your sdk_config.h

TissieVA, 2021-12-14:
[image]

After turning these on, the error still remains unfortunately.

[image]

TomRaeves, 2021-12-16:
After playing with the enables and checking the includes it is still not working.

MatsDeMeyer, 2021-12-16:
Fixed now? What did you change?

TissieVA, 2021-12-16:
It seems that a line in our Makefile had escaped our realm and had been hidden in the shadows of the universe, sending errors upon us.  😉

TomRaeves, 2021-12-16:
Maybe for everyone else wanting to use BLE uart. You have to look at the uart example and copy paste uart_init and uart_event_handle in your code. After this you have to call the uart_init in your main. Then the last 2 things you have to do is add the makefile includes that were in the example in your own project and lastly you have to set some things to 1 in de sdkconfig.h the same as in the example again.

closed 2021-12-16 by MatsDeMeyer