MOSAIC-LoPoW/octa-stack-students #55: Interrupts already defined

MatthiasW1004, 2022-01-25:
If I try to generate an interrupt using a push button, I can't make the bin file because the EXTI0_IRQHandler is already defined elsewhere. How can I make it use my IRQHandler? 
[image]
[image]

MatsDeMeyer, 2022-01-25:
The clean solution is using the EXTI0_IRQHandler that is already defined in the stm32l4xx_it file, and placing your code there.
If you really want to use your own handler, delete the already defined one or make it "weak" so that it uses another one if there is one

MatthiasW1004, 2022-01-25:
I fixed it like this:
[image]

MatthiasW1004, 2022-01-25:
For the RTC alarm callback, I want to use HAL_RTC_RegisterCallback. It is defined in the stm32l4xx_hal_rtc file but it does not recognize the defined parameters. 
[image]
[image]

MatsDeMeyer, 2022-01-25:
VSCode does tend to highlight stuff as undefined, does your code compile?

MatthiasW1004, 2022-01-25:
No it does not, it gives me the 'not declared' error

Klievan, 2022-01-26:
Could you post the full error the compiler is giving?

MatthiasW1004, 2022-01-26:
[image]

MatthiasW1004, 2022-01-26:
I am also wondering why the code where this function is defined is more blurry than the code above: 
[image]

MatthiasW1004, 2022-01-26:
Do you maybe have an example in your octastack about how to use the RTC-Alarm interrupt? Or something similar?

Klievan, 2022-01-26:
Hi Matthias,

Mats is on vacation but I'll try to help you out here. Like you noticed, the function that you're trying to call is dimmed, which means that the #if directive above is probably returning false, so the code below is not being included. Try defining USE_HAL_RTC_REGISTER_CALLBACKS to 1 in your makefile
https://stackoverflow.com/questions/53288150/how-to-add-define-in-makefile

MatthiasW1004, 2022-01-26:
I  changed the value in the hal_conf_template file but it still does not work. For some reason I also cannot add the define in the make file as said in the stackoverflow discussion

Klievan, 2022-01-26:
Can you post the line you added in your makefile?

Klievan, 2022-01-26:
Are you getting the same error?

MatthiasW1004, 2022-01-26:
[image]
And I get the same error. I also tried changing it here: 
[image]

MatthiasW1004, 2022-01-26:
[image]
This is the documentation about it

Klievan, 2022-01-26:
And the error remains the same? Have you tried to do a clean build?

MatthiasW1004, 2022-01-26:
Yes, tried the clean build, even tried other ways to define the USE_HAL_RTC_REGISTER_CALLBACKS but still the same error.

Klievan, 2022-01-26:
Did you import ./core/ST/STM32L4xx_HAL_Driver/Inc/stm32l4xx_hal_rtc.h?

MatthiasW1004, 2022-01-26:
[image]
I did like this, I think it is correct right?

Klievan, 2022-01-26:
It is if you also added the correct entries in your makefile. Could you show those?

MatthiasW1004, 2022-01-26:
[image]

This is the full make file