MOSAIC-LoPoW/octa-stack-students #56: Cutting solder bridge SB40 of the Power Profiler Kit

WouterHuygen, 2022-01-25:
Is it required to cut solder bridge SB40 on the nRF52840 DK for making measurements with the PPK on external DUTs?
This is mentioned in the documentation: 
https://infocenter.nordicsemi.com/index.jsp?topic=%2Fug_nrf52840_dk%2FUG%2Fdk%2Fprepare_board.html

Klievan, 2022-01-25:
Hi! Are you trying to measure the current of the application running on the nRF52840 dev-kit, yes. If you are measuring an external DUT, I'm 99% sure you can skip that step.

WouterHuygen, 2022-01-25:
I'm measuring on an external DUT.
However, when I turn on the timestamps, the measurements seem to be not exactly 'live' or realtime and I was wondering what causes this.
Below are my results when measuring on the 3.3v and ground pin while sending LoRa messages. But when I remove the LoRa module, I don't see any change in current. So I am assuming I am doing something wrong, but I have no idea what it could be...
[image]

My PPK is configured as follows:
- DUT select: External
- Power select: Reg
- COM: DK

Klievan, 2022-01-25:
Hi Wouter!

What is your setup electrically? Are you able to show a picture or draw a simple schematic how you connected everything together?

Klievan, 2022-01-25:
So you are powering the OCTA through a USB cable which means the Nordic PPK isn't able to monitor the power usage. Disconnect the octa USB cable and power it through the Nordic PPK. There is another issue that I believe already has been closed which explains how to connect the OCTA to the PPK

WouterHuygen, 2022-01-25:
Ah yes, indeed that was my mistake. Now my measurements make a lot more sense. Thank you!
[image]

closed 2022-01-26 by WouterHuygen