MOSAIC-LoPoW/octa-stack-students #57: Listening to UART using IWDG

TomRaeves, 2022-01-26:
We cant seem to figure out how we can listen to the UART because we are using IWDG and so our functions only gets called periodically after certain time points so we will miss some information if it gets sent inbetween these times. Is there a way to continuously listen to the UART after pressing a button for an amount of time when we are using the periodical IWDG timers for everything else ? 
@MatsDeMeyer @michielaernouts

TomRaeves, 2022-01-26:
It is working if we enable a timer which listens on the UART every 0.1 seconds but I don't know if this is the most elegant/foolproof way to do it.

SamGeens, 2022-01-26:
It is possible to use UART with interruption. This will call an interrupt handler every time a UART message is received. I've not done this myself so I can't help with that but at least it is possible. I guess that if you're looking for the most elegant way this is the one.

TomRaeves, 2022-01-26:
i dont think its possible to use an interrupt on the UART if you are trying to decide when to listen to the UART.

SamGeens, 2022-01-26:
I'm afraid I don't get the point. 

> our functions only gets called periodically after certain time points so we will miss some information if it gets sent inbetween these times

You said you're missing information because UART is not handled because other code is being executed? The whole point about the interrupt is that it only fires when something is received on UART. So if I'm right you don't need to decide when to listen on UART.

If not I might misunderstood your question :) Some officials should take it from here I think ;)

jeroen1323, 2022-01-26:
I think Sam is right. 
If you use  `HAL_UART_Receive_IT(&UART2, &UART2_rxBuffer, 1);` instead of  `HAL_UART_Receive(&UART2, &UART2_rxBuffer, 1);` you will get an interrupt when you receive a message over UART. So by adding a corresponding callback function you should be able to continue. When this interrupt is triggered the scheduled tasks will be on hold for as long as the code is in the callback, so you would not miss any messages. 
However, do not use the `HAL_UART_Receive_IT` in a loop, calling it once is enough to enable the UART interrupt.

SamGeens, 2022-01-26:
@jeroen1323 `HAL_UART_Receive_IT` is indeed the function I couldn't remember!